tue-robotics-graveyard/robot_smach_states #11: Manipulation: automatically determine which arm to use

svddries, 2015-01-08:
Now, if one arm is broken, a challenge / demo fails if the wrong arm 'side' is specified in the executive. We should have a way of determining which arm is available and then use that arm.

LoyVanBeek, 2015-01-08:
This is more complicated: If an arm fails *after* grasping something, its not possible to place that object somewhere else. This can still fail. 
After grasping, the correct arm for a place action is the arm that is holding that item. 

I guess we can make some designators that do this: they resolve to an Arm-object.

Rayman, 2015-01-27:
We could do the following:

- for each side of the robot
  - if a grasp succeeds, `occupied=true`
  - if a place succeeds, `occupied=false`
  - if something fails, `occcupied=false`

LoyVanBeek, 2015-01-27:
That could be a start, but we also want to know which arm is holding what object. 
So: 
- for each side of the robot
  - if a grasp succeeds, `arm/occupied=<entity_ID>`
  - if a place succeeds, `occupied=None`
  - if something fails, `occcupied=None`

We can then even add a method to the robot-class that returns a list of empty arms and the arm (if any) that is holding some entityID.

LoyVanBeek, 2015-01-30:
The occupied_by should be a property instead of a normal member so it can encapsulate a a procedure to modify the entity's TF and set its parent to be the gripper frame.

LoyVanBeek, 2015-02-11:
This is being used in challenge_manipulation. Needs full testing though.

closed 2015-02-11 by LoyVanBeek