tue-robotics-graveyard/robot_smach_states #14: All designators should resolve to Entities rather than EntityIDs

LoyVanBeek, 2015-01-31:
Some states use designators that resolve to the *ID* of an ED Entity.
Other states use designators that resolve to actual ED Entities.
We must standardize to the 2nd option, because getting an Entity's ID when you have the Entity is easier than the other way around; then you must do a query, which takes time. 

Some examples: 
- manipulation.SetGripper
- a bunch of navigation states
- etc?

You can use 
     
    id_designator = AttrDesignator(entity_designator, 'id')

for conversion purposes by the way

jlunenburg, 2015-02-10:
I agree. However, we need to do this transition quickly in order not to break to much, preferably before the upcoming workshop. In addition, I added the argument 'parse' to the 'get_entities' function. In the library experiment, this could take up to 6 (!!!) seconds while most of the time the additional info that is obtained by parsing is not required.

LoyVanBeek, 2015-02-10:
I wil start this standardization tonight. I think most use cases are in navigation, although I'm not sure.

LoyVanBeek, 2015-02-10:
Merged to master in b42145d0b6e3aa29aba48c9432cd24f4c69f5ad3

closed 2015-02-10 by LoyVanBeek