tue-robotics-graveyard/robot_smach_states #15: Designators in say state

jlunenburg, 2015-02-04:
The following code in the Say state sometimes crashes (line 56). Will post a terminal output if I have one:
-        try:
-            #Resolve designators if there are any in the format
-            def resolve_if_desig(fmt):
-                if isinstance(fmt, Designator):
-                    return fmt.resolve()
-                else:
-                    return fmt
-            concrete_format_args = map(resolve_if_desig, self.format_args)
-            concrete_format_kwargs = {key:resolve_if_desig(value) for key, value in self.format_kwargs.iteritems()}
-            sentence = sentence.format(*concrete_format_args, **concrete_format_kwargs)
-        except DesignatorResolvementError, e:
-            rospy.logerr(e)
-            pass

LoyVanBeek, 2015-02-04:
Crap, @reinzor replaced Say with SayFormatted yesterday in commit https://github.com/tue-robotics/robot_smach_states/commit/ac240dd8e91d33804ea381796abf375f7ad5a031 
If you revert that commit, it should be fine again.

reinzor, 2015-02-04:
> SayFormatted yesterday in commit ac240dd

The other way around ..?

-Rein

LoyVanBeek, 2015-02-04:
Yesterday, we replaced the Say-state with SayFormatted and renamed it to Say. So the API of Say changed an causes breakage apparently.

closed 2015-02-14 by LoyVanBeek