tue-robotics-graveyard/robot_smach_states #30: Crash on designator resolve

LFBFerreira, 2015-04-01:
Yesterday it was working, today its not, so some recent change broke it.

When trying to resolve a designator i get this output:

[ERROR] [WallTime: 1427900176.466746] InvalidUserCodeError: Could not execute state 'LOOK_AT_OPERATOR' of type '<person_recognition_states.LookAtPersonInFront object at 0x7f4a6990b210>': Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/smach/state_machine.py", line 247, in _update_once
    self._remappings[self._current_label]))
  File "/home/luis/ros/indigo/repos/https:/github.com/tue-robotics/tue_robocup.git/challenge_person_recognition/src/person_recognition_states.py", line 136, in execute
    desgnResult = humanDesignator.resolve()
  File "/home/luis/ros/indigo/system/src/robot_smach_states/src/robot_smach_states/util/designators/designator.py", line 323, in resolve
    entities = self.ed.get_entities(_type, _center_point, self.radius, _id, self.parse)
  File "/home/luis/ros/indigo/system/src/robot_skills/src/robot_skills/world_model_ed.py", line 44, in get_entities
    center_point = self._transform_center_point_to_map(center_point)
  File "/home/luis/ros/indigo/system/src/robot_skills/src/robot_skills/world_model_ed.py", line 92, in _transform_center_point_to_map
    point_in_map = transformations.tf_transform(pointstamped.point, pointstamped.header.frame_id, "/map", self._tf_listener)
  File "/home/luis/ros/indigo/system/src/robot_skills/src/robot_skills/util/transformations.py", line 110, in tf_transform
    output_coordinates = tf_listener.transformPoint(outputframe, ps)
  File "/home/luis/ros/indigo/system/src/tf_server/src/tf_server/tf_client.py", line 38, in transformPoint
    raise tf.Exception(resp.error_msg)
Exception: Invalid argument passed to lookupTransform argument source_frame in tf2 frame_ids cannot be empty

jlunenburg, 2015-04-01:
Already mentioned this in another issue

LFBFerreira, 2015-04-01:
I was not sure which package is creating the crash, so i put it here. Should i delete it or wait to see where is the problem?

LoyVanBeek, 2015-04-01:
Oh boy I really broke it bad... I fix this asap

LoyVanBeek, 2015-04-01:
I can reproduce this error in the amigo-console:

    >>> from robot_skills.util.msg_constructors import *
    >>>  from robot_smach_states.util.designators import *
    >>> EdEntityDesignator(amigo, id="avoid_that_start").resolve() 
   Exception ....

LoyVanBeek, 2015-04-01:
The tiniest diff...

commit e5324c2b (2015-04-01): Point instead of PointStamped :-(. Fixed #30
closed 2015-04-01 by LoyVanBeek