tue-robotics-graveyard/robot_smach_states #40: VariableDesignators: Make clear which states sets the variable and which only read them

LoyVanBeek, 2015-08-04:


rokusottervanger, 2015-08-25:
Do you mean by naming conventions?

LoyVanBeek, 2015-08-26:
Nope, was working on making variable designators unwritable unless explicitly marked as writeable.

rokusottervanger, 2015-08-26:
What exactly would you do with an unwritable variable designator?

LoyVanBeek, 2015-08-26:
Just get data from it, as you would with any designator. But to write to it, the writing state has to have the variable explicitly as writeable. 
In your code, you can see exactly which states then write to any VariableDesignator and which just read. 
Suggestions welcome.

rokusottervanger, 2015-08-26:
So when one state needs to write to a designator, and a second one needs to read that information, the designator should be writable, right? If that's the case, could you give an example of a situation where a designator should not be writable?

LoyVanBeek, 2015-08-26:
Well, the reader does not need to know that a designator is a VariableDesignator or whatever. Only the writer must know that it can write, and should check that it is writeable. 
Readers should only check the resolve_type.

closed 2015-09-29 by LoyVanBeek