tue-robotics-graveyard/robot_smach_states #42: Designators are not needed

Rayman, 2015-08-19:
Why Designators are not needed
--------------------
In short: just use bigger skills or bigger smach dates. Two examples of behaviors that are solved by Designators but actually have a better solution:

Passing data between states
- userdata (already exists)

A generic state which uses some input to do its work
- userdata
- or dont have a generic state but just a generic skill

Designators are too complicated
--------------------
They uglify the APIs when they are used. They also uglify the code *when* you have to use them because everyting needs to be wrapped.

Unneeded Designators (all of them)
--------------------

ArmDesignator & UnoccupiedArmDesignator
- just remember which arms are occupied (already there) and add a function to arms.py for selecting arms

ArmHoldingEntityDesignator
- for each arm remember what is occupying it (already there) and add a function to arms.py for selecting arms

ListElementDesignator
- is not used

Designator
- use a variable

VariableDesignator
- this is a communication channel, so use one or use userdata

PointStampedOfEntityDesignator
- belongs in skills

EdEntityCollectionDesignator
- belongs in skills

EdEntityDesignator
- belongs in skills

EmptySpotDesignator
- belongs in skills

LockingDesignator (used for locking an arm)
- if you use bigger smach states you wont run into this problem

AttrDesignator
- is not used

FuncDesignator
- is not used

DeferToRuntime
- is not used

LoyVanBeek, 2015-08-20:
I would like to agree, but experience indicates that the alternatives we used (userdata and asserting/querying data to/from the reasoner) also suck quite badly.

The problem with Smach userdata is that composing states and statemachines together is not as simple as we would like. 
When setting some userdata in a state that is embedded in several layers of state machines, getting it out somewhere else in your application is not simple, nontransparent and very error prone. You can have userdata-keys with the same name etc. 

The purpose for designators is mostly to encapsulate and reuse the code to deal with ED, the reasoner when we still used it, userdata and the things that can go wrong there etc. 

When we used the reasoner, the code to get answers, convert that to a usable data type, do calculations and filtering etc was duplicated all over the place. That is what designators solved, because now you just have a .resolve() call that gives you a usable answer directly. You may pass in a custom designator, defined in your application/RoboCup-challenge and don't have to change the statemachines. 

We now also use it as a sort of dataflow-graph which makes things more complex but also more powerful. 
The point is that sometimes you want to control, at the application level, exactly what a state deeply embedded in the state hierarchy does. So, you send someone to do that for you, but you need to instruct that someone.
Designators allow you to separate the how something is done (what states do) from the with-what something is done (designators) Yes, you can let states decide with what to do something, but then your application has less or no control over that. And you duplicate the code to make that decision. 

As to the usecases of all the designators above:
- ArmDesignators: indeed, they wrap the functionality in arms.py. You can give a state an ArmDesignator that returns either an unoccupied arm or an arm holding some item, buth with the same interface. The alternative is to pass to the state a bool/ID to select an arm based on it. Then the logic to do this is duplicated and you loose the ability to do more advanced stuff specifically for your application.  

- ListElementDesignator: I expected to use this as a base class for others. 
- Designator: yes you can use a variable, but then you must always use a variable and you loose the ability to make things dynamic or decided at runtime. 
- VariableDesignator: userdata sucks, really. Try using it for something non-trivial, you'll bump into problems. 
- PointStampedOfEntityDesignator: should be deleted indeed
- EdEntityCollectionDesignator: Can be a skill, but you loose the ability to do anything dynamic or put control in your application
- EdEntityDesignator: Wraps a skill mostly but adds some filtering functions. Again, you want to have control over e.g. which type of object you want to select. You can pass that type to the state calling the skill, but then you have to pass in more and more arguments to that state, through the hierarchy.  
- EmptySpotDesignator: Could be a skill, but I can also pass a different designator that also returns a spot, but implemented a completely different way. I can choose which to use now at the application level if I want to. 
- LockingDesignator. For arms indeed, using larger pieces of code that claim the arm is OK, but it is also used for EdEntities, to commit to using that Entity.
- For the other ones, I expected use-cases and they were easy enough to implement in a few lines.

Rayman, 2015-08-26:
I've made an example of three states where userdata is used to pass information from a `hear` state to a `query` state and I could then also add a manipulation state. I looks really simple and I'm planning to include more examples.

https://github.com/tue-robotics/robot_smach_states/blob/master/examples/hear_grasp.py

LoyVanBeek, 2015-08-26:
Yes, of course that is possible in this simple example. But what happens when I want to GetEntitiesBy<somethingelse>? And not just get the closest one? And other application-specific things. 
Then, you will need more and more code in the GetEntitiesBy...-state. If you want to do that again somewhere else, you need to duplicate the code. To get that to do exactly what you want, you need to parametrize that logic and pass those via the state's constructor which then gets ugly. 

An alternative to duplicating that code it to put in in some common place and let that output the result. 

I've been here before. Believe me and @jlunenburg if we say that userdata sucks for anything complex and reusable.

closed 2015-09-29 by LoyVanBeek