tue-robotics-graveyard/robot_smach_states #50: Designator crash when no entity can be resolved

reinzor, 2016-02-17:
https://github.com/tue-robotics/robot_smach_states/blob/master/src/robot_smach_states/util/designators/core.py#L209 failes when list is empty

[ERROR] [WallTime: 1455718713.116732] InvalidUserCodeError: Could not execute state 'SEGMENT' of type '<robot_smach_states.world_model.SegmentObjects object at 0x7effc829d290>': Traceback (most recent call last):
  File "/home/amigo/ros/indigo/system/src/smach/src/smach/state_machine.py", line 247, in _update_once
    self._remappings[self._current_label]))
  File "/home/amigo/ros/indigo/system/src/robot_smach_states/src/robot_smach_states/world_model.py", line 103, in execute
    self.objectIDsDes.write(objClassif)
  File "/home/amigo/ros/indigo/system/src/robot_smach_states/src/robot_smach_states/util/designators/core.py", line 209, in write
    if isinstance(value[0], self.variable_designator.resolve_type[0]):
IndexError: list index out of range

LoyVanBeek, 2016-02-23:
Added a check for this case, please test

Rayman, 2016-09-17:
Is this still an issue?

closed 2016-12-06 by LoyVanBeek