tue-robotics-graveyard/robot_smach_states #52: Place and grasp poses seem to be wrong

reinzor, 2016-02-23:
Using robot_smach_states.Grab()

[image]
[image]
[image]

LoyVanBeek, 2016-02-23:
We need to clean up the various grab-states we have an use/optimize ONE

reinzor, 2016-03-02:
Issues solved:

- [ ] Entity gets updated before grasp

Some issues still left:

- [ ] Grasp pose still seems to be wrong if I check visualization
- [ ] Get rid of SjoerdsGrab

Rayman, 2016-09-17:
Did we get rid of SjoerdsGrab?

closed 2016-09-17 by Rayman