tue-robotics-graveyard/robot_smach_states #53: Why don't we use the carrying pose as reset pose?

reinzor, 2016-03-16:
@MaxBaeten 
@jlunenburg 

It always takes much time to go up and prepare for a grasp, what is the downside of making the reset configuration the carying pose? Except for looks maybe?

MaxBaeten, 2016-03-16:
Is that really a considerable amount of time? And those movements can't they be executed in parallel to looking for the to be grasped object?

Hardwarewise there aren't really objections to this besides that it might not look that nice while driving.

reinzor, 2016-03-16:
Looking for the object is really fast. The movements are taking most of the
time. It is just a little bit weird that we always go back to the reset
position and then to the same sequence again. We can look at this Saturday
:).

-Rein

MaxBaeten, 2016-03-16:
That is true, best is to have the arm in reset position while driving but already going to grasp mode much earlier. But I am still wondering why it takes so long. Is it actually moving the arm all that time or does is also take a long time before the arm starts moving. i.o.w. would increasing the velocity of the arm help (Don't think there is a lot to gain there, at least safely :) )

reinzor, 2016-03-16:
We will discuss saturday :). But yes, it takes long :/

-Rein

Rayman, 2016-09-17:
This is an option for #147

closed 2017-02-28 by jlunenburg