tue-robotics-graveyard/robot_smach_states #59: Grab machine crash

reinzor, 2016-10-29:
Executive hangs:

        [INFO] [WallTime: 1477743413.484479] State machine terminating 'STOP_ANALYSIS_SUCCEED':'done':'arrived'
        [INFO] [WallTime: 1477743413.484607] State machine transitioning 'NAVIGATE_TO_GRAB':'arrived'-->'GRAB'
        [WARN] [WallTime: 1477743417.567100] Not waiting for gripper action
        Succesfully stored '/home/amigo/ed/kinect/2016-10-29/67f2149afcdb70f402a6df4971b9b131/2016-10-29-14-16-58_rgb.png'
        [INFO] [WallTime: 1477743418.752666] ID to update: 67f2149afcdb70f402a6df4971b9b131
        [ERROR] [WallTime: 1477743418.752841] Could not resolve the updated grab_entity, this should not happen [CHECK WHY THIS IS HAPPENING]
        GPD took 0.005716 seconds
[ERROR] [WallTime: 1477743418.782262] InvalidUserCodeError: Could not execute state 'GRAB' of type '<robot_smach_states.manipulation.grab.PickUp object at         0x7f4d40129990>': Traceback (most recent call last):
          File "/home/amigo/ros/indigo/system/src/smach/src/smach/state_machine.py", line 247, in _update_once
            self._remappings[self._current_label]))
          File "/home/amigo/ros/indigo/system/src/robot_smach_states/src/robot_smach_states/manipulation/grab.py", line 147, in execute
            tf_listener=self.robot.tf_listener)
          File "/home/amigo/ros/indigo/system/src/robot_skills/src/robot_skills/util/transformations.py", line 128, in tf_transform
            output_coordinates = tf_listener.transformPoint(outputframe, ps)
          File "/home/amigo/ros/indigo/system/src/tf_server/src/tf_server/tf_client.py", line 38, in transformPoint
            raise tf.Exception(resp.error_msg)
        Exception: "998549f4cee3049842157e5e31e55eee" passed to lookupTransform argument source_frame does not exist. 
        
        [ERROR] [WallTime: 1477743418.782562] State 'GRAB' failed to execute.
        [ERROR] [WallTime: 1477743418.782733] State 'SELF_CLEANUP' failed to execute.
        [ERROR] [WallTime: 1477743418.782871] State 'HANDLE_DETECTED_ENTITIES_0' failed to execute.
        [ERROR] [WallTime: 1477743418.783008] State 'INSPECT_0' failed to execute.
[ERROR] [WallTime: 1477743418.783161] Robot exited with <class 'smach.exceptions.InvalidUserCodeError'>,Could not execute state 'GRAB' of type '<        robot_smach_states.manipulation.grab.PickUp object at 0x7f4d40129990>': Traceback (most recent call last):
          File "/home/amigo/ros/indigo/system/src/smach/src/smach/state_machine.py", line 247, in _update_once
            self._remappings[self._current_label]))
          File "/home/amigo/ros/indigo/system/src/robot_smach_states/src/robot_smach_states/manipulation/grab.py", line 147, in execute
            tf_listener=self.robot.tf_listener)
          File "/home/amigo/ros/indigo/system/src/robot_skills/src/robot_skills/util/transformations.py", line 128, in tf_transform
            output_coordinates = tf_listener.transformPoint(outputframe, ps)
          File "/home/amigo/ros/indigo/system/src/tf_server/src/tf_server/tf_client.py", line 38, in transformPoint
            raise tf.Exception(resp.error_msg)
        Exception: "998549f4cee3049842157e5e31e55eee" passed to lookupTransform argument source_frame does not exist. 
        ,<traceback object at 0x7f4d406fe050>
        [INFO] [WallTime: 1477743418.783536] Torso cancelling all goals on close
        [INFO] [WallTime: 1477743418.783767] left arm cancelling all goals on all arm-related ACs on close
        [INFO] [WallTime: 1477743418.784561] right arm cancelling all goals on all arm-related ACs on close
        Traceback (most recent call last):
          File "/home/amigo/ros/indigo/system/src/r5cop_demo/src/challenge/amigo/challenge_amigo.py", line 52, in <module>
            robot_smach_states.util.startup(setup_statemachine, challenge_name="r5cop_demo_amigo")
          File "/home/amigo/ros/indigo/system/src/robot_smach_states/src/robot_smach_states/util/startup.py", line 78, in startup
            outcome = executioner.execute()
          File "/home/amigo/ros/indigo/system/src/smach/src/smach/state_machine.py", line 359, in execute
            container_outcome = self._update_once()
          File "/home/amigo/ros/indigo/system/src/smach/src/smach/state_machine.py", line 250, in _update_once
            raise ex
smach.exceptions.InvalidUserCodeError: Could not execute state 'GRAB' of type '<robot_smach_states.manipulation.grab.PickUp object at 0x7f4d40129990>': Traceback (most         recent call last):
          File "/home/amigo/ros/indigo/system/src/smach/src/smach/state_machine.py", line 247, in _update_once
            self._remappings[self._current_label]))
          File "/home/amigo/ros/indigo/system/src/robot_smach_states/src/robot_smach_states/manipulation/grab.py", line 147, in execute
            tf_listener=self.robot.tf_listener)
          File "/home/amigo/ros/indigo/system/src/robot_skills/src/robot_skills/util/transformations.py", line 128, in tf_transform
            output_coordinates = tf_listener.transformPoint(outputframe, ps)
          File "/home/amigo/ros/indigo/system/src/tf_server/src/tf_server/tf_client.py", line 38, in transformPoint
            raise tf.Exception(resp.error_msg)
        Exception: "998549f4cee3049842157e5e31e55eee" passed to lookupTransform argument source_frame does not exist. 
        
Middleware:

        [ERROR] [1477743418.780844779]: Transform error at 2nd attempt

LoyVanBeek, 2016-11-01:
How do we want to solve this issue? 
It seems to be due to a frame not yet or no longer existing frames. 

What to do if a frame does not exist is up to the application code. For the GrabMachine, signalling that the object went missing is the only thing left to do, and then go look for it again in the application (robocup challenge or demo)

LoyVanBeek, 2016-11-01:
Further fixes are in the tf_client/server that now allow to wait for a transform: https://github.com/tue-robotics/tf_server/commit/39a9c6b6a04d42a743559169b5c47608d365a33e

reinzor, 2016-11-09:
I don't think this problem is solved:

        Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 122 in /tmp/binarydeb/ros-kinetic-tf2-0.5.13/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 122 in /tmp/binarydeb/ros-kinetic-tf2-0.5.13/src/buffer_core.cpp

closed 2017-02-28 by jlunenburg