tue-robotics-graveyard/robot_smach_states #60: Navigate to symbolic 'near' is not working

reinzor, 2016-11-09:
Near area is calculated wrong, could also be a fault in ed_navigation but this has to be checked!

LoyVanBeek, 2016-11-09:
In what way is it not working?

reinzor, 2016-11-09:
Wrong area is calculated 

-Rein

LoyVanBeek, 2016-11-09:
Which challenge, designator/entity, what area name?

reinzor, 2016-11-09:
see r5cop challenge

-Rein

LoyVanBeek, 2016-11-12:
Can also be tested with 
```bash
rosrun robot_smach_states simple_navigate.py amigo
```

If you change the object in https://github.com/tue-robotics/robot_smach_states/blob/master/examples/simple_navigate.py#L45 to 'plant' or 'trashbin' and 'near', the robot goes near them, so that works.
It also works when going 'near' the 'nightstand' or 'bar'.
It does not work for 'bookcase' or 'cabinet'. 

plant and trashbin are simple cylinders and the nightstand and the bar are boxes. 
bookcase and cabinet are compound shapes, so I think the convex hull of these may be incorrect.

closed 2016-11-12 by LoyVanBeek

reinzor, 2016-11-12:
> Closed #60 via tue-robotics/ed_navigation@d0f259d.

Nice!

-Rein