tue-robotics-graveyard/robot_smach_states #61: Remove all direct dependencies on packages, let everything flow via the robot_skills layer

reinzor, 2016-11-23:
Everything should go via robot_skills, not directly talking with services or actionlibs, this is spaghetti

LoyVanBeek, 2016-11-23:
Note to self:

- [ ] ed_perception is referenced via https://github.com/tue-robotics/robot_smach_states/search?utf8=%E2%9C%93&q=ed_perception and https://github.com/tue-robotics/robot_smach_states/search?utf8=%E2%9C%93&q=FaceLearningGoal, so can be removed AFAIK
- [ ] ed_navigation is referenced only in the package.xml so can also be removed: https://github.com/tue-robotics/robot_smach_states/search?utf8=%E2%9C%93&q=ed_navigation&type=Code

reinzor, 2016-11-23:
This should I think also include all messages that are being used in the robot smach states. However, this will break a lot most probably :(

LoyVanBeek, 2016-11-23:
It makes sense for this package to directly depend on a bunch of msgs, most clearly 

- cb_planner_msgs_srvs.msg
- std_msgs
- geometry_msgs
- ed
- visualization, for debugging. 

These should be included in the CMakelists.txt rather than inheriting these from robot_skills.

reinzor, 2016-11-23:
I don't think so, the robot_skills layer should never pass something ROS specific. IMO, robot_smach_states can even be ROS agnostic

closed 2017-02-28 by jlunenburg